Triton-Robotics/TR-mbed #15: Is motor online

pknessness, 2023-08-02:
Have a very simple function/variable that is a reflection of if a motor is online (has it been giving us feedback) or not.

GioByte10, 2023-08-02:
```C++
if(motorsExist[bus][r][c]) {
    if (us_ticker_read() / 1000 - lastCalled[bus][r][c] > TIMEOUT_MS) {
        if(debug) {
            printf("Motor %d on bus: %d lost connection\n", c + 4 * r + 1, bus + 1);
            printf("Motor[%d][%d][%d]\n", bus, r, c);
        }
        return false;
    }
}
```

pknessness, 2023-08-03:
yeah ok thats what it was, i remmeber that you had done something similar but i realized during comp it would be more useful to have it motor specific so i can be like oh `if(pitch.isConnected)` or smth

GioByte10, 2023-08-11:
https://github.com/Triton-Robotics/TR-mbed/commit/75d873e79ab2e41fc0a973d05f26d90ec1a50106

closed 2023-09-26 by GioByte10